eldendiss/DendoStepper #19: Expand async usage possibility

cedricwritescode, 2023-12-06:
**Is your feature request related to a problem? Please describe.**
When starting a movement, it will get executed asynchronously (using the gptimer ISR). To check that the movement is completed, however, you need to poll the status.

**Describe the solution you'd like**
The library should tell the main program that the motor movement finished, by means of a callback of some sorts. Maybe the esp_event library could be used.

**Describe alternatives you've considered**
FreeRTOS constructs could be used as well, like binary Semaphores, queues or task notifications.

**Additional context**

cedricwritescode, 2023-12-06:
[This issue](https://github.com/eldendiss/DendoStepper/issues/20) needs to be fixed for this enhancement.

dolfelt, 2024-10-06:
Are there any updates on this? I am trying to get feedback from a motor movement from one ESP32 running the motor, to another ESP32 running an LCD touchscreen.

cedricwritescode, 2024-10-07:
Together with @eldendiss I tried improving the architecture (#22). I think, however, nothing in the async direction was actually implemented. Currently I'm quite busy with other stuff so maybe he can help out, or just fork this library :)