eldendiss/DendoStepper #26: TROUBLE MAKING IT WORK WITH TI MOTOR DRIVER

v-3n0m, 2024-02-02:
**Describe the bug**
Hello there I was trying to test the example with the TMC2208 but not able to start the stepper motor.

But a simple gpio control works

The delay between the the high and low pulse should be 250 ms 

**HW (please complete the following information):**
 - ESP32 module: [ESP32S3]
 - Stepper driver: [TMC2208]

digiexchris, 2024-02-02:
I personally would be really tempted to use UART with that chip, instead of bit banging step/dir/en. it looks like it really nicely integrates with an MCU that way, including a CRC for error correcetion! Very nice. 

Do you have any oscilloscope captures of the step pin? There seem to be a lot of configurable options on that driver that can affect things. https://www.digikey.ca/en/htmldatasheets/production/2042923/0/0/1/mikroe-2624

I don't think a constant like 250ms is a good idea, because that limits the entire library to a max speed of not more than 120 pulses per second. Besides, that driver's datasheet (Section 11. Timing) indicates that the minimum pulse width is 100 nanoseconds, not 250 milliseconds. 250 milliseconds is 4 per second, VERY slow. Can you share a minimum code sample that duplicates the issue?

[image]

v-3n0m, 2024-02-06:
Thank you for your reply  I found the issue .. The issue was I was using the gpio number 42,35 in my s3 somehow it was not generating enough pulses changed the gpio to different one and there it is working